matthuszagh/pyems #3: Installation instructions

thomaslepoix, 2020-08-13:
Hey! Your tool looks great, but not being used to python, I can't manage to install it. Here is what I have done. (Ubuntu 18.04)

```sh
cd /usr/local/src/
sudo git clone https://github.com/thliebig/openEMS-Project --recurse-submodules
sudo git clone https://github.com/matthuszagh/pyems

cd /usr/local/src/openEMS-Project/
sudo ./update_openEMS.sh /usr/local --with-hyp2mat --with-CTB --with-MPI

cd /usr/local/src/openEMS-Project/CSXCAD/python/
sudo python3 setup.py install

cd /usr/local/src/openEMS-Project/openEMS/python/
sudo python3 setup.py install

cd /usr/local/src/pyems/
sudo python3 setup.py install
```

So if I run an example script, I got this error : 

```sh
pyems/examples/ $
	./microstrip.py

		Traceback (most recent call last):
		  File "./microstrip.py", line 4, in <module>
			from pyems.structure import PCB, Microstrip
		  File "/usr/local/lib/python2.7/dist-packages/pyems-0.1.0-py2.7.egg/pyems/structure.py", line 25
			prop: CSProperties,
				^
		SyntaxError: invalid syntax
```

So I changed the first line from `#!/usr/bin/env python` to `#!/usr/bin/env python3`, and then :

```sh
pyems/examples/ $
	./microstrip.py

		Traceback (most recent call last):
		  File "./microstrip.py", line 4, in <module>
			from pyems.structure import PCB, Microstrip
		  File "/usr/local/lib/python3.6/dist-packages/pyems-0.1.0-py3.6.egg/pyems/structure.py", line 11, in <module>
		  File "/usr/local/lib/python3.6/dist-packages/CSXCAD/__init__.py", line 7, in <module>
			from CSXCAD.CSXCAD import ContinuousStructure
		ImportError: libCSXCAD.so.0: cannot open shared object file: No such file or directory

	locate libCSXCAD.so.0
		/usr/local/lib/libCSXCAD.so.0
		/usr/local/lib/libCSXCAD.so.0.6.2
```

The library is present on my system, I probably missed something. Do you have an idea?

Thank you in advance. :)

matthuszagh, 2020-08-14:
Thanks for filing an issue! This looks like a problem with the openems python interface. Try importing `ContinuousStructure` directly from python. I.e.

1. first run:
```
$ python3
```

2. then at the python console run:
```
>>> from CSXCAD.CSXCAD import ContinuousStructure
```

And let me know if you see the same shared library error.

thomaslepoix, 2020-08-14:
Exactly the same error.

matthuszagh, 2020-08-14:
Ok, so this is indeed an issue with how you've installed csxcad, rather than with pyems. It's been a while since I've used the normal python installation process (I use Nix) so I'm a bit rusty on the details, but I'm setting up a VM with Ubuntu now so I'll be able to test this/debug your issue. In the meantime, have you tried using `pip`? And I'd drop most of the `sudo` stuff, I don't think you need it.

So, for instance I'd do something like:
```
mkdir -p ~/src
cd ~/src
git clone https://github.com/thliebig/openEMS-Project --recurse-submodules
git clone https://github.com/matthuszagh/pyems
cd openEMS-Project
sudo ./update_openEMS.sh /usr/local --with-hyp2mat --with-CTB  # drop mpi, i've had issues with it with python
cd CSXCAD/python
pip install .
cd ../../openEMS/python
pip install .
cd ../../../pyems
pip install .
```

Then try the same import.

matthuszagh, 2020-08-14:
Also, what's your `LD_LIBRARY_PATH` set to?
```
echo $LD_LIBRARY_PATH
```

matthuszagh, 2020-08-14:
Ok, I was able to get this to work. The direct issue you're facing is that python looks to `LD_LIBRARY_PATH` for your library location, but since it's not set (I'm assuming yours isn't set, bc when I setup my Ubuntu VM it was not set) it can't find it. The superficial fix for this is what I said above: set it when running ./microstrip.py. But, the better thing to do is to specify the library dir when installing. If you're using pip you can do:

```
pip install --global-option="build_ext" --global-option="-l/usr/local/lib" ~/src/openEMS-Project/CSXCAD/python
```

See [this SO answer](https://stackoverflow.com/a/22942120/5710525) for reference.

You're going to experience another issue after you've resolved this one related to importing `SmoothMeshLines.py`. There's currently a bug in upstream CSXCAD. I've [submitted a patch](https://github.com/thliebig/CSXCAD/pull/19) to fix it, so if you want to have your install working right away, either edit it manually, or use my branch to install.

Next, pyems requires that you install AppCSXCAD. Just follow the directions in the INSTALL file if you haven't already. Also get paraview if you want to view field dumps.

Finally, this is a stylistic recommendation, but unless you have a reason not to, I'd be installing everything with `pip` and in a virtual environment (see [here](https://docs.python.org/3/tutorial/venv.html)). I'd also avoid `sudo` (to reiterate my recommendation above) unless you're doing that for a specific reason.

thomaslepoix, 2020-08-15:
Thank you for your informations and your reactivity!

No MPI, ok.

I finally solved the problem reinstalling both openEMS and CSXCAD with `-R`. I precise changing flags does not trigger a new build, one have to manually remove the `build` directory not to spend hours trying to figure out what happens.

```sh
sudo -H pip install --global-option="build_ext" --global-option="-R/usr/local/lib/" --prefix=/usr/local .
```

All the `sudo` and the `--prefix=/usr/local` are optional of course. It is a personal preference because I think `/usr/local/` (and `/usr/local/src/` for sources) are fine. I dislike installations in home directories, especially in a case like this one, where the python thing to install is a part of a system wide installed software. 

You're right, I'm new to python

I suggest you to set the shebangs of example scripts to `python3` since those are incompatibles with older versions and I think the user shouldn't have to use venv or to change its default settings in order to run scripts.

closed 2020-08-15 by thomaslepoix

matthuszagh, 2020-08-15:
Glad you were able to get it working. I'm also agreed with you in regard to the shebang lines. I've fixed this in the [latest commit](efff93ce124540ac6bd1a85f1370b9a77bad7519).