matthuszagh/pyems #18: allow use of lorentz model for thin metals

matthuszagh, 2020-09-15:
See the [openems documentation](https://openems.de/index.php/Dispersive_Material_Property.html) for the lorentz model.

matthuszagh, 2020-09-19:
See [this thread](http://openems.de/forum/viewtopic.php?f=3&t=131), which contains a lot of useful information.

biergaizi, 2023-06-30:
Another related improvement is the use of Lorentz model for PCB dielectrics to model frequency-dependent permittivity and dispersion. The difficult here is the lack of data of the wideband measurement of PCB materials. But according to this overview, [*Modeling frequency-dependent dielectric loss and dispersion for multi-gigabit data channels*](https://www.simberian.com/AppNotes/ModelingDielectrics_2008_06.pdf), apparently the Djordjevic-Sarkar model is common for electronics design. Its robustness has been confirmed in multiple IEEE and DesignCon papers. The best advantage is that it allows the use of dielectric constant and loss tangent as few as a single frequency point only, while still gives reasonable results.

So a potential path of modeling PCBs is:

1. Input: dielectric constant, loss tangent, and frequency, at one or more points.
2. Apply the Djordjevic-Sarkar model to find the permittivity curve.
3. Use numerical optimization to find the parameter for a high-order Lorentz material.
4. Pass the result as material definition to openEMS.