matthuszagh/pyems #23: mesh option to ignore very small structures

matthuszagh, 2020-10-17:
This is useful when your model inadvertently creates tiny structures that would increase the simulation time dramatically. This effect crops up with miter structures, which are used for example in the [coupler simulation](https://github.com/matthuszagh/pyems/blob/master/examples/coupler.py), shown below.

[image]

In this image the effect isn't terrible, but it could be much worse if the miter was configured in a slightly different way causing the inner corner to have almost the same x-position as the upper right corner. Probably, the best solution to this will be to add a parameter for mesh generation to "ignore" structures below a certain size (given in terms of the minimum simulation wavelength). For these small structures, I'll probably place one mesh line midway between the bounds so the structure is still somewhat registered and leave it at that. This could present problems when two very small structures are adjacent. I can't think of an instance when this would happen, but it could break this solution so it's worth considering.

thomaslepoix, 2020-10-20:
Hey!

It seems to me the thirds rule is applied in the wrong direction for 3 of the 4 metal / non-metal orthogonal edges. Also it seems thirds rule is applied to the 2 metal / metal edges which does not make sense and may be the origin of the too close mesh lines here.

matthuszagh, 2020-10-20:
Yep, you're correct. The current third's rule behavior is indeed wrong, and it's due to [this issue](https://github.com/matthuszagh/pyems/issues/2). I need to think a little bit about the best way to solve this. The reason is a bit of technical one based on the way pyems handles meshing, but unfortunately it will crop up a lot with flat microstrip simulations (which is of course a major use case) until I fix it.

As for the close mesh lines, I think you're referring to the 2 locations where the miter meets the microstrips? This issue is a bit more complicated. I previously opened and closed an [issue for this](https://github.com/matthuszagh/pyems/issues/9). This is kind of a special case. Specifically, the miter has the same initial width as the microstrip. I could identify this and merge these `BoundedType`s so they get treated as a single `BoundedType`. However, if the miter had a different width than the microstrip (this wouldn't happen with a miter, but it very well might with another structure) it would make sense to keep them as separate.

I think the best solution to this problem is to merge adjacent structures if one structure is basically a continuation of the other. In this case, the bounding box of the miter region to the right of the inner corner is an exact continuation of the bounding box of the microstrip, so it makes sense here.

Reopening #9.