matthuszagh/pyems #25: Be able to return complex values from s_param function in Simulation Class

laberge1, 2020-11-05:
Today, only magnitude can be read out of the s_param function within the Simulation Class, despite setting the "dB" input bool to False.  I am sure there are ways around this, but it seems limiting to not allow phase information to be passed out.  I propose that by setting the "dB" input to "False" should return the complex value, versus the magnitude/absolute value.  This is important for building your own touchstone file (which would also be a nice function to have, but easy enough to build using scikit-rf).

matthuszagh, 2020-11-05:
Closing this in favor of your [PR](https://github.com/matthuszagh/pyems/pull/26). I reverted my merge for that because of the commit message. But once you change the commit message please recreate the PR and I'll merge it.

closed 2020-11-05 by matthuszagh

laberge1, 2020-11-05:
Opened a [PR](https://github.com/matthuszagh/pyems/pull/30) with a helpful commit message.