matthuszagh/pyems #35: Miter example doesn't work

clemc98, 2021-05-27:
Hi,
I'm experiencing an issue when using the Miter example that come with pyems.
```
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-1-f4986749549a> in <module>
     26 )
     27 
---> 28 miter = Miter(
     29     pcb=pcb,
     30     position=Coordinate2(pcb_len / 4, pcb_len / 4),

~/.local/share/virtualenvs/install_7-trBMitfD/lib/python3.8/site-packages/pyems-0.1.0-py3.8.egg/pyems/structure.py in __init__(self, pcb, position, pcb_layer, gnd_layer, trace_width, gap, rotation, miter, transform)
   1987 
   1988         if self.position is not None:
-> 1989             self.construct(self.position)
   1990 
   1991     @property

~/.local/share/virtualenvs/install_7-trBMitfD/lib/python3.8/site-packages/pyems-0.1.0-py3.8.egg/pyems/structure.py in construct(self, position, transform)
   2041         self._position = c2_maybe_tuple(position)
   2042         self._construct_trace()
-> 2043         self._construct_gap()
   2044 
   2045     def _construct_trace(self) -> None:

~/.local/share/virtualenvs/install_7-trBMitfD/lib/python3.8/site-packages/pyems-0.1.0-py3.8.egg/pyems/structure.py in _construct_gap(self)
   2078         )
   2079         construct_polygon(
-> 2080             prop=prop,
   2081             points=self._gap_points(),
   2082             elevation=self.pcb.copper_layer_elevation(self._pcb_layer),

NameError: name 'prop' is not defined
```

matthuszagh, 2021-05-27:
Yeah a number of these examples are unfortunately broken as I'm still changing the API regularly. I'll take a look. Thanks for reporting.

matthuszagh, 2021-05-28:
Should be fixed with 55b9d34163e10273018e39974f16943c1e67b885. I think this example could probably be optimized a bit since the interesting part (i.e., miter) is physically small compared to the full structure. That seems suboptimal. The solution would likely be to make the metal resolution finer and decrease the pcb dimension. This is really a guess though as I haven't tested it.

closed 2021-05-28 by matthuszagh