matthuszagh/pyems #46: microstrip_sma_transition.py: RuntimeError: PML mesh smoothing is increasing the simulation box size.

biergaizi, 2022-03-15:
I was trying to the demo `microstrip_sma_transition.py`, unfortunately it fails.

```
$ python3 microstrip_sma_transition.py
/usr/lib64/python3.9/site-packages/scipy/optimize/minpack.py:175: RuntimeWarning: The iteration is not making good progress, as measured by the 
  improvement from the last five Jacobian evaluations.
  warnings.warn(msg, RuntimeWarning)
/usr/lib64/python3.9/site-packages/scipy/optimize/minpack.py:175: RuntimeWarning: The iteration is not making good progress, as measured by the 
  improvement from the last ten iterations.
  warnings.warn(msg, RuntimeWarning)
Traceback (most recent call last):
  File "/home/user/code/pyems/examples/microstrip_sma_transition.py", line 216, in <module>
    mesh = Mesh(
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py", line 594, in __init__
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py", line 619, in generate_mesh
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py", line 722, in _smooth_pml_mesh_lines
RuntimeError: PML mesh smoothing is increasing the simulation box size.  This should never happen.  Either structures in your PML are non-uniform in the direction of the PML in which case you'll get an error anyway, or this is a bug in pyems.
```

matthuszagh, 2022-03-15:
This runs ok for me. What revision of CSXCAD are you running? This particular example requires a patched version of CSXCAD. Unfortunately, the [PR](https://github.com/thliebig/CSXCAD/pull/15) has not been accepted (yet). So maybe that's the issue?

biergaizi, 2022-03-15:
Ah, thanks. This indeed could be the issue here!

matthuszagh, 2022-03-15:
I'll add a note in the example stating this. It doesn't look like I actually mentioned that anywhere.

biergaizi, 2022-03-15:
Perfect. I've applied the patch, now it runs correctly.

It still generates some meshing warnings though, which I assume is harmless.

```
$ python3 microstrip_sma_transition.py
/usr/lib64/python3.9/site-packages/scipy/optimize/minpack.py:175: RuntimeWarning: The iteration is not making good progress, as measured by the 
  improvement from the last five Jacobian evaluations.
  warnings.warn(msg, RuntimeWarning)
/usr/lib64/python3.9/site-packages/scipy/optimize/minpack.py:175: RuntimeWarning: The iteration is not making good progress, as measured by the 
  improvement from the last ten iterations.
  warnings.warn(msg, RuntimeWarning)
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos 5.4672 for dimension 0 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.21. For convenience the last three lines are: 5.2833, 5.3667 and 5.4672.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos -3.5886 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.25. For convenience the last three lines are: -3.7875, -3.6770 and -3.5886.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos -2.5030 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.31. For convenience the last three lines are: -2.5641, -2.5294 and -2.5030.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos -2.3888 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.32. For convenience the last three lines are: -2.4500, -2.4236 and -2.3888.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos -0.6189 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.20. For convenience the last three lines are: -0.7935, -0.6982 and -0.6189.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos -0.2360 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.35. For convenience the last three lines are: -0.2679, -0.2496 and -0.2360.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos -0.1778 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.30. For convenience the last three lines are: -0.2090, -0.1954 and -0.1778.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos 0.2090 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.30. For convenience the last three lines are: 0.1778, 0.1954 and 0.2090.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos 0.2679 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.35. For convenience the last three lines are: 0.2360, 0.2496 and 0.2679.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos 0.7935 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.20. For convenience the last three lines are: 0.6189, 0.6982 and 0.7935.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos 2.4500 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.32. For convenience the last three lines are: 2.3888, 2.4236 and 2.4500.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos 2.5641 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.31. For convenience the last three lines are: 2.5030, 2.5294 and 2.5641.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos 3.7875 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.39. For convenience the last three lines are: 3.5684, 3.6600 and 3.7875.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos -2.3780 for dimension 2 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.32. For convenience the last three lines are: -2.4393, -2.4044 and -2.3780.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos -2.2639 for dimension 2 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.31. For convenience the last three lines are: -2.3250, -2.2986 and -2.2639.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos -1.1468 for dimension 2 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing w [... cut ...]
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos 2.5750 for dimension 2 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.32. For convenience the last three lines are: 2.5136, 2.5486 and 2.5750.
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py:664: UserWarning: Mesh line at pos 2.6893 for dimension 2 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.32. For convenience the last three lines are: 2.6280, 2.6544 and 2.6893.
```

closed 2022-03-15 by biergaizi

matthuszagh, 2022-03-15:
Great, glad it works. Those warnings are indeed harmless. They indicate that the mesher wasn't able to satisfy the stated smoothness factor, though the ~1.3 it's achieving really isn't bad IMO.

matthuszagh, 2022-03-15:
FYI: 2a2139329b83623d9efd489d1241a7fd9d5e5207.