matthuszagh/pyems #49: gcpw_bypass_cap.py: TypeError: __init__() got an unexpected keyword argument 'via_gap'

biergaizi, 2022-03-15:
I was trying to run the demo gcpw_bypass_cap.py, unfortunately it fails.

```
Traceback (most recent call last):
  File "/home/user/code/pyems/examples/gcpw_bypass_cap.py", line 33, in <module>
    Microstrip(
TypeError: __init__() got an unexpected keyword argument 'via_gap'
```

biergaizi, 2022-03-15:
I looked at another working example `gcpw_blocking_cap.py`, and I found the via-related code is commented out, including Via Gap and Via Wall. Does it mean that the via simulations in these examples are no longer compatible and cannot be used anymore in the current codebase?

matthuszagh, 2022-03-15:
I decided to remove these as parameters to `Microstrip` and add dedicated classes for via walls/fences in 338497abb27d56e557301db855a3bfe9355e1de6. These would need to be instantiated separately. So, yes this interface is no longer supported (and is a bug in any examples that use it), but the same functionality is still possible.