matthuszagh/pyems #51: Reducing Upper Frequency Range Causes Meshing Failures (RuntimeError: Probe or feed overlaps PML)

biergaizi, 2022-03-16:
I've noticed a peculiar behavior when exploring the example `gcpw_blocking_cap.py`. By default, the simulation frequency range is:

    req = np.arange(0, 18e9, 1e7)

And pyems is able to generate a simulation model correctly. However, I'm only interested in the high-frequency behavior up to 10 GHz, thus I've change it to

    req = np.arange(0, 10e9, 1e7)

Now it's unable to generate a model anymore due to PML overlap.

```
Traceback (most recent call last):
  File "/home/user/code/pyems/examples/gcpw_blocking_cap.py", line 134, in <module>
    mesh = Mesh(
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py", line 594, in __init__
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py", line 624, in generate_mesh
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/simulation.py", line 221, in post_mesh
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/simulation.py", line 229, in _mesh_errors
RuntimeError: Probe or feed overlaps PML. Please fix your simulation. CSX file has been saved so you can view the overlap.
```

I tried tweaking the parameter `feed_shift` on the excitation port but it doesn't seem to make any difference.

matthuszagh, 2022-03-16:
> I guess reducing the frequency causes the meshing algorithm to use a reduced resolution, which affected the correct placement of PML.

This is correct, though I'm not sure I'd refer to it as placement of the PML. The PML will occupy the last 8 cells regardless of the cell size, but will obviously change in total extent depending on that cell size. I've attached an image showing the excitation within the PML. It's on the left side, in the grey boundary (the PML), just below the top conductor of the microstrip. You can see the same thing yourself by running `AppCSXCAD sim/gcpw_blocking_cap.xml`. You'll need to toggle off visibility of some parts of the structure in the AppCSXCAD GUI to see the excitation and probes. Also worth mentioning that these grey boxes aren't actually the PML. I've added them so that it's easier to see where the PML is, but they have no effect on the simulation.

The resolution would either be to (1) make the cells smaller, (2) make the structure larger in the direction of the microstrip, or (3) (as you've identified) shift the excitation and probes. I'd probably go with option 2 here.

When you used `feed_shift` you may have moved the feed out of the PML but not the probes. This would be done with `measurement_shift`.

[image]

biergaizi, 2022-03-16:
Thanks for your answer and patience, especially the hint on `measurement_shift`. 

P.S: due to the complexity of E&M simulation and the lack of tutorials on OpenEMS in general, I think I (or anyone who's trying out this framework) am going to ask many questions in the future, including open-ended questions on modeling strategies. I don't think GitHub Issue is the best place for Q&A. Perhaps creating a *GitHub Discussions* forum is good idea.

matthuszagh, 2022-03-17:
> Perhaps creating a _GitHub Discussions_ forum is good idea.

I think that's a good idea - it's now created. Eventually, I'd like all of that info to make its way into the documentation, but I think discussions could also be a good staging area for that info.

biergaizi, 2022-03-18:
I'm closing this one as well, since it has been answered.

closed 2022-03-18 by biergaizi

warlock31415, 2024-07-20:
> make the structure larger in the direction of the microstrip
@matthuszagh  what do you mean by this? which structure are you talking about?