matthuszagh/pyems #54: Allow different material properties for different substrate layers

matthuszagh, 2022-03-17:
Raised [here](https://github.com/matthuszagh/pyems/issues/39#issuecomment-1067748041).

biergaizi, 2022-03-18:
The dielectric layers are easy to code. But I think the concerning issue is the solder mask filling problem.

If the solder mask is modeled just as a uniform layer on both substrate and copper, the simplest case, we can simply fill it as a uniform layer and use the OpenEMS priority system to let copper take priority.

But in fact, the solder mask thickness on top of a copper trace is different from the solder mask on a bare substrate. For example, the most popular commercial model, Si9000, models this effect, and it's often the official model used by many PCB manufacturers. 

[image]

If we want to model this...

1. It means the software must be able to identify the copper region and substrate region and to apply the mask on a case-by-case basis. 

2. A more tricky problem is the solder mask filling problem - on the edge of a copper trace, there will be exposed copper. For example, if the copper is 1.40 mils, solder mask on top of copper is 0.50 mils, and solder mask on substrate is 0.80 mils, it means there will be 0.6 mils of exposed copper on the edge of the substrate solder mask. This transition region should be also filled by solder mask. In my personal experiment, TNT-MMTL (a forgotten but likely the most powerful FOSS 2D transmission line field solver), is able to reproduce the result of Si9000 for a differential pair to within 1 ohms if the edge solder mask is modeled. However, if it's not modeled, there will be a difference of a few ohms.

However, there's also the question of whether it's worthwhile to model that in a 3D E&M solver. I think a 3D full-wave solver like OpenEMS is not really suitable for modeling such small features. Unlike a 2D solver, OpenEMS is more sensitive to simulation artifacts. Even on a bare board with a single dielectric layer, there will be a error of a few ohms if the meshing is just a little bit off. Not to mention that the difference lies within the normal PCB manufacturing tolerance.

matthuszagh, 2022-03-23:
This is good commentary. And thanks for pointing out TNT-MMTL. It's always good to compare results in pyems/openems to results from other simulators.

There are a few additional complications/thoughts I'll mention:

1. Pyems uses [`AddConductingSheet`](https://openems.de/index.php/Metal_Property.html) for PCB microstrips, etc. Unlike `AddMetal`, conducting sheets are 2-dimensional. This makes the simulation run faster (we don't have small non-zero extent structures) and should be more accurate than `AddMetal`, which assumes perfect conductors. IIRC the math behind conducting sheets is based on a paper that I can probably dig up if you're interested. But, this doesn't really work with physical structure of the solder mask over trace in the figure above. Openems [supports the Drude-Lorentz model](https://openems.de/index.php/Drude/Lorentz_Material_Property), which I've been wanted to investigate for a while, but haven't yet had the chance. This could be one area where that's useful.
2. Another thing I've had on my todo list is to implement a tolerance analysis for pyems. Since the electrical properties of soldermask and physical properties of its application are poorly controlled, I suspect this will have a larger effect on measured quantities than small physical details about the structure. But this is just a guess, and I think these more accurate geometric models are certainly worth exploring (and the two don't need to be mutually exclusive). A tolerance analysis applies to a lot of other things too. An obvious example is when simulating FR4. But even when using better materials, the manufacturer tolerances for trace widths, prepreg thickness, etc. can have a significant effect on simulation results.
3. If we're discussing soldermask, it's probably also worth considering finishes. For example, in some respects ENIG finishes are worse than soldermask (see [1](https://gitlab.com/dlharmon/rf-blocks/-/tree/main/LRL_cal/OSHPark4L_cal) and [2](https://harmoninstruments.com/posts/oshpark_4l.html)). Modelling ENIG accurately would require a more sophisticated metal model (Drude-Lorentz might work here).

Finally, I think it's worth mentioning that when we get into details like this it becomes difficult to validate the results the simulator gives us. I can order a PCB and perform various S-parameter and impedance measurements, but I don't know of a way to measure soldermask or prepreg thickness. It may be that the best approach is to validate against a high-quality commercial simulator that may perform these validations. Alternatively, maybe it would be possible to simulate certain very simple structures with a high degree of accuracy and computation cost (i.e., very fine meshes, accurate metal models, etc.) and use that as a reference for faster, less accurate simulations. But, that's only a rough idea at this point.

biergaizi, 2023-02-23:
> IIRC the math behind conducting sheets is based on a paper that I can probably dig up if you're interested. But, this doesn't really work with physical structure of the solder mask over trace in the figure above.

Why do you say that the math of conducting sheets doesn't really work with solder mask coating? I think solder mask is nothing special but it's just a dielectric layer. If it doesn't work, does it mean that conducting sheets also don't work as inner layers of PCB? How does pyEMS model the ground planes in a multi-layer PCB then? As lossless metal?

I'm started to think that using a "virtual" dielectric layer may work as a workaround to simulate the effect of solder mask. That is, it doesn't have to physically model the actual solder mask, we can just tweak its parameter (i.e. dielectric constant) until the trace impedance matches the experimental value. Thus, we can bypass the problem of modeling the extremely thin layer of coating, which is problematic for FDTD.

matthuszagh, 2023-02-23:
> Why do you say that the math of conducting sheets doesn't really work with solder mask coating? I think solder mask is nothing special but it's just a dielectric layer. If it doesn't work, does it [...]

Oh, I simply meant that by flattening the trace you're also necessarily changing the structure of the soldermask. Whether or not that really matters, I don't know.

> I'm started to think that using a "virtual" dielectric layer may work as a workaround to simulate the effect of solder mask. That is, it doesn't have to physically model the actual solder mask, we can just tweak its parameter (i.e. dielectric constant) until the trace impedance matches the experimental value (or 2D solver value). Thus, we can bypass the problem of modeling the extremely thin layer of coating, which is problematic for FDTD.

I'm not totally sure I follow. Are you suggesting to create a thicker dielectric layer for which you modify it's attributes until the simulation results are similar to what they would have been with the thinner, accurately-modeled layer? I suppose you could run some tests and try to do that, but I expect it would be tough to find something that is generally accurate. My guess is that it would be model-specific and probably pretty band-limited, but I don't know.

I wonder how much it really makes sense to angst about small effects from soldermask though. Even if you could model it accurately, I'm not aware of any manufacturing process that can apply it with a very precise and even thickness. I don't think the dielectric is generally very well controlled either. I've been under the impression that if you really need that level of precision, you should probably use a different way of protecting the copper (immersion silver maybe?).

biergaizi, 2023-02-23:
> I'm not totally sure I follow. Are you suggesting to create a thicker dielectric layer for which you modify it's attributes until the simulation results are similar to what they would have been with the thinner, accurately-modeled layer?

Yes.  In one simple experiment, I was trying to model the effect of a shunt SMD component. I found the only reason that the simulation disagrees with experimental data is due to the difference of characteristic impedance caused by solder mask. It would be convenient if the geometry in a simulation can be "nudged" to match experiments, without additional adjustments like tweaking trace width.

I tried to run the IEEE P370 de-embedding algorithm with a 2xThru line to remove the effect of the virtual fixture like how you would do it with a real VNA, and I found the S-parameters are almost completely identical after this process. But it feels pretty inconvenient.

Using a dielectric layer to "nudge" the characteristic impedance sounds like a useful idea for some kinds of simulations.

> I wonder how much it really makes sense to angst about small effects from soldermask though.

I think the concept of systematic error and random error is useful here. Solder mask is poorly controlled, so it has large production variations, which can be seen as a random error. If tight impedance control is needed, solder mask should be removed. But we can be pretty sure that overall it systematically adds a downward bias to the characteristic impedance in all manufactured boards. If openEMS is used for parameter-sweep type optimizations, ideally the result of the computation should stay at the center of this distribution. This is why 2D solvers like Si9000 tries to model it.

matthuszagh, 2023-03-09:
Sorry for the delay in getting back to you.

> I think the concept of systematic error and random error is useful here. Solder mask is poorly controlled, so it has large production variations, which can be seen as a random error. If tight [...]

This is a good point and I agree.

> Yes. In one simple experiment, I was trying to model the effect of a shunt SMD component. I found the only reason that the simulation disagrees with experimental data is due to the difference of characteristic impedance caused by solder mask. It would be convenient if the geometry in a simulation can be "nudged" to match experiments, without additional adjustments like tweaking trace width.
> 
> I tried to run the IEEE P370 de-embedding algorithm with a 2xThru line to remove the effect of the virtual fixture like how you would do it with a real VNA, and I found the S-parameters of simulation and experimental data are almost completely identical after this process. But it feels pretty inconvenient.
> 
> Using a dielectric layer to "nudge" the characteristic impedance sounds like a useful idea for some kinds of simulations.

I think it certainly makes sense to support soldermask layers for PCB structures. And, with that in place, it wouldn't be hard for users to adjust thicknesses and dielectric constants to their liking. And we can certainly document results for popular stackups that we found to work well for various results like characteristic impedance. But, I wouldn't want to provide physically unrealistic defaults unless we had a pretty high level of confidence that it generalized well. Still, I think it's an intriguing idea and if you have any models/measurements etc you want to share I'd be curious to hear them.

On a related note (and I think we might have discussed this a little while back), I think the `PCBProperties` and maybe `PCB(Structure)` code in pyems is too rigid and should be re-written somewhat. I haven't looked at the code in a little while (so I may be overlooking/misremembering things), but it would probably make more sense if PCBs were just a collection of layers, each of which had a type and properties, as well as via properties, which don't belong to a single layer. In this context soldermask layers would be trivial to add as they would just be dielectric layers on the bottom and top. This would be much more flexible, and you could do seemingly funny things like putting two metal layers on top of each other, but I don't see any problem with that.

This doesn't address your point, but it's also worth being a bit skeptical of any simulations with SMD components. What OpenEMS does with this is very much a simplification/approximation (i.e., it doesn't model internal structures), and I never felt like I got very reliable results with it. In theory, this could be done in a more physically accurate way but that information often isn't available and even if it was would probably lead to unacceptably long simulation times. IIRC sonnet handles this by not actually simulating the passive and instead using measured S-parameter data for it. There might be a good reason for that.

altavo-dominik, 2025-02-17:
Is there any update on implementing custom dielectric layers? I would like to give your api a try but without being able to model the pcb accurately with different cores and prepregs it doesn't seem very useful to me. All the PCBs I manufacture are a mix of Rogers and various prepregs with sometimes significantly differents dielectric constants. If you explain how you want it implemented I could probably make you a PR..

biergaizi, 2025-03-16:
> Is there any update on implementing custom dielectric layers? I would like to give your api a try but without being able to model the pcb accurately with different cores and prepregs it doesn't seem very useful to me.

To clarify, there are several problems for accurate PCB modeling:

1. One must model different layers of dielectric materials, with different permittivity (dielectric constant). The current pyems code only models one layer.

    Multi-layering is fairly easy to implement, doable in a weekend I'd say. Basically, the current constructor of class `PCBProperties` is defined like this:

    ```
    PCBProperties(
            substrate=project_dielectrics["Generic-FR-4"],
            copper_thickness=[0.0356e-3, 0.0356e-3],
            substrate_thickness=[1.6e-3],
            metal_conductivity=5.8e7,
            via_plating_thickness=0.020e-3,
    )
    ```

    The parameter `substrate` only accepts a single value, such as:

        Dielectric(epsr=[(1, 4.6)], loss=[(1, 0.03)]),

    To implement multi-layering, one needs to rewrite the code so parameter `substrate` accepts a Python list of object `Dielectric` (not just a single `Dielectric`). Each `Dielectric` class should also record its own thickness instead of relying on the `substrate_thickness` parameter.

    Furthermore, from a code clarify point of view, the proposed change may be potentially objectionable, since `Dielectric` should only represent an abstract material, not a concrete layer with thickness. So perhaps as an alternative solution, one can define a new class called `Stackup` that can be constructed from a `Dielectric` with an additional thickness?

    At any rate, the change should be fairly simple, doable with a weekend.

    But there are additional barriers that prevent accurate simulations.

2. Very thin dielectric layers (such as solder masks) are very difficult to calculate in FDTD.

    Thus they can't be modeled without using some tricks I mentioned, e.g. ignore the solder mask, and tweak the permittivity of the substrate until it matches the effect of the solder mask - which is not a rigorous solution. But since the effect would usually be small, so it won't be a problem in most cases. However, I've reported that in some extreme cases, such as closely-spaced coplanar waveguide, the effect of the dielectric is very significant. It's enough to break most empirical formulas, only a field solver that models this thin layer can do it properly. We have a discussion at here:

    * [Modeling Strategy for Solder-Mask Coated Microstrip Traces?](https://github.com/thliebig/openEMS-Project/discussions/38). It showed that while it's impractical to model a full circuit board, a single, isolated transmission line can still be modeled by brute-force. 

3. The current code assumes material permittivity is constant, so the simulation result is only accurate at one frequency. But permittivity is actually a function of frequency, not a constant. So in order to be accurate throughout all frequencies, one must use the Drude-Lorentz material model, with parameters fitted to a circuit board substrate. We have a discussion at here:

    * [Some Simulation Results with Wideband Dielectric models](https://github.com/thliebig/openEMS-Project/discussions/157) Basically the problem is threefold.

    1. First, no accurate wideband circuit substrate data is available, permittivity is only given at one frequency. Solution: Common circuit board characteristics can be extrapolated from a single data point via Djordjevic-Sarkar model.
    2. The frequency-dependent permittivity must be converted to a format usable by openEMS's Debye-Lorentz model. Solution: A Matlab script is written to find a numerical fit. 

    The remaining problem is that the Debye-Lorentz code has extremely low performance, I have some ideas about where the bottleneck is and how to solve it, but it requires rewriting the code. Also, numerical stability has also been reported.

4. Surface roughness losses are not modeled. openEMS has two conductor models, perfect electric conductor, and lossy thin conductor sheet. But the sheet model only models resistive losses, not surface roughness losses. We have a discussion at here:

    * [Conductor loss for wideband models](https://github.com/thliebig/openEMS-Project/discussions/161)

5. Lumped S-parameters are not available. Currently openEMS is not able to represent an arbitrary lumped component defined by its S-parameters.

    > This doesn't address your point, but it's also worth being a bit skeptical of any simulations with SMD components. What OpenEMS does with this is very much a simplification/approximation (i.e., it doesn't model internal structures), and I never felt like I got very reliable results with it.

    This is why it doesn't work. We had a discussion at here about lumped inductance (which has been solved), but lumped S-parameters is still an open problem. See [RLC series vs parallel + inductor support?](https://github.com/thliebig/openEMS-Project/discussions/65#discussioncomment-6278803)