matthuszagh/pyems #55: [PATCH] feed_impedance attribute is broken (AttributeError: module 'numpy' has no attribute 'is_complex')

biergaizi, 2022-03-17:
I was trying to set the `feed_impedance` parameter in pyems, unfortunately it triggers a Python exception.

```
#!/usr/bin/env python3

import numpy as np
from pyems.pcb import common_pcbs
from CSXCAD.CSTransform import CSTransform
from pyems.simulation import Simulation
from pyems.structure import (
    PCB,
    Microstrip,
)
from pyems.mesh import Mesh
from pyems.coordinate import Axis

unit = 1e-3
freq = np.arange(0, 18e9, 1e7)
sim = Simulation(freq=freq, unit=unit)
pcb_prop = common_pcbs["oshpark4"]
pcb_len = 10
pcb_width = 5

pcb = PCB(
    sim=sim,
    pcb_prop=pcb_prop,
    length=pcb_len,
    width=pcb_width,
    layers=range(3),
    omit_copper=[0],
)

microstrip1 = Microstrip(
    pcb=pcb,
    position=(0, 0),
    length=1,
    width=1,
    propagation_axis=Axis("x"),
    trace_layer=0,
    gnd_layer=1,
    port_number=1,
    excite=True,
    feed_impedance=50,
)

mesh = Mesh(
    sim=sim,
    metal_res=1 / 120,
    nonmetal_res=1 / 40,
    smooth=(1.5, 1.5, 1.5),
    min_lines=5,
    expand_bounds=((0, 0), (8, 8), (8, 8)),
)
```

Running this demo triggers the exceptions below.

```
Traceback (most recent call last):
  File "/home/user/code/bugfix/pyems/pyems/bug.py", line 43, in <module>
    mesh = Mesh(
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py", line 594, in __init__
    self.generate_mesh()
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py", line 624, in generate_mesh
    self.sim.post_mesh()
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/simulation.py", line 219, in post_mesh
    self._align_ports_to_mesh()
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/simulation.py", line 317, in _align_ports_to_mesh
    [port.snap_to_mesh(mesh=self.mesh) for port in self.ports]
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/simulation.py", line 317, in <listcomp>
    [port.snap_to_mesh(mesh=self.mesh) for port in self.ports]
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/port.py", line 112, in snap_to_mesh
    self._set_feed(mesh=mesh)
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/port.py", line 541, in _set_feed
    feed = Feed(
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/feed.py", line 55, in __init__
    self.set_feed()
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/feed.py", line 88, in set_feed
    rval, cval, lval = self._impedance_rcl()
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/feed.py", line 116, in _impedance_rcl
    if np.is_complex(self.impedance):
  File "/usr/lib64/python3.9/site-packages/numpy/__init__.py", line 214, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'is_complex'
```

A quick look at the code suggests `is_complex` is a typo. But after correcting this typo, it still triggers another exception.

```
/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/structure.py:1111: UserWarning: Reference impedance not set for port 1
Traceback (most recent call last):
  File "/home/user/code/bugfix/pyems/pyems/bug.py", line 43, in <module>
    mesh = Mesh(
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py", line 594, in __init__
    self.generate_mesh()
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/mesh.py", line 624, in generate_mesh
    self.sim.post_mesh()
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/simulation.py", line 219, in post_mesh
    self._align_ports_to_mesh()
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/simulation.py", line 317, in _align_ports_to_mesh
    [port.snap_to_mesh(mesh=self.mesh) for port in self.ports]
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/simulation.py", line 317, in <listcomp>
    [port.snap_to_mesh(mesh=self.mesh) for port in self.ports]
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/port.py", line 112, in snap_to_mesh
    self._set_feed(mesh=mesh)
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/port.py", line 541, in _set_feed
    feed = Feed(
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/feed.py", line 55, in __init__
    self.set_feed()
  File "/home/user/.local/lib/python3.9/site-packages/pyems-0.1.0-py3.9.egg/pyems/feed.py", line 98, in set_feed
    prop=res, box=self.box, priority=priorities["component"],
NameError: name 'priorities' is not defined
```

This problem appears to be caused by a missing import in feed.py.

After applying this patch, both problems are fixed, though I can't be sure whether it's fully functional either...

```patch
diff --git a/pyems/feed.py b/pyems/feed.py
index 196a857..5b370e2 100644
--- a/pyems/feed.py
+++ b/pyems/feed.py
@@ -4,6 +4,7 @@ import numpy as np
 from pyems.simulation import Simulation
 from pyems.mesh import Mesh
 from pyems.utilities import max_priority
+from pyems.priority import priorities
 from pyems.coordinate import Box3, box_overlap
 from pyems.csxcad import construct_box
 
@@ -113,7 +114,7 @@ class Feed:
     def _impedance_rcl(self) -> Tuple[float, float, float]:
         """
         """
-        if np.is_complex(self.impedance):
+        if np.iscomplex(self.impedance):
             warn("Only feed resistances are currently supported.")
 
         return (np.real(self.impedance), 0, 0)
```

matthuszagh, 2022-03-18:
This looks correct. Feel free to make a PR and I'll accept.

biergaizi, 2022-03-18:
I just opened PR #56 for this.