matthuszagh/pyems #57: ModuleNotFoundError: No module named 'CSXCAD' (Ubuntu 20.04.4)

0xCoto, 2022-04-08:
I have [built](https://github.com/BlitPlatform/compute-servers/blob/main/build_solver.sh) OpenEMS successfully, and everything works fine from Octave on Ubuntu 20.04.4. However, when I try cloning:
```bash
git clone https://github.com/matthuszagh/pyems
```
and then running the `horn_antenna.py` example:
```bash
cd pyems/examples
cp horn_antenna.py .. # Moving one dir back so it identifies the pyems directory
python3 horn_antenna.py
```
I get this error:
```python
Traceback (most recent call last):
  File "horn_antenna.py", line 6, in <module>
    from pyems.utilities import print_table
  File "/home/apostolos/Desktop/oems_tests/pyems/pyems/utilities.py", line 5, in <module>
    from CSXCAD.CSXCAD import ContinuousStructure
ModuleNotFoundError: No module named 'CSXCAD'
```

Any ideas if I should try to install something in a different manner?

Thanks in advance!

matthuszagh, 2022-04-08:
That means the [CSXCAD python bindings](https://github.com/thliebig/CSXCAD/tree/master/python) aren't installed (or at least not visible to python). Are you building csxcad in addition to openems?

FYI this and most other examples will run fine, but the microstrip to coax transition example requires [a patch to csxcad](https://github.com/thliebig/CSXCAD/pull/15), which has not yet been merged upstream.

0xCoto, 2022-04-08:
@matthuszagh Thank you for your quick reply! Indeed, I had forgotten to add `./python3-openems*.deb` to the build script. It now runs OK, but I'm facing a few other issues:

When I try to import STL files like so:
```py
# Patch
patch = sim.csx.AddMetal("patch")
stl1 = patch.AddPolyhedronReader(
    filename=os.path.abspath("/home/apostolos/Desktop/Part__Feature001.stl")
)
stl1.ReadFile()

# Substrate
substrate = sim.csx.AddMaterial("substrate", epsilon=2.2, mue=1, kappa=0, sigma=0)
stl2 = substrate.AddPolyhedronReader(
    filename=os.path.abspath("/home/apostolos/Desktop/Part__Feature002.stl")
)
stl2.ReadFile()

# Ground
ground = sim.csx.AddMetal("ground")
stl3 = ground.AddPolyhedronReader(
    filename=os.path.abspath("/home/apostolos/Desktop/Part__Feature.stl")
)
stl3.ReadFile()
```

I get the following window popup:
```
An geometry read error occurred!!

Warning: Invalid primitive found in property: patch!
Warning: Invalid primitive found in property: substrate!
Warning: Invalid primitive found in property: ground!
```

As far as I know, this is often caused by the file not being found by the script, but I've specified the full path (which does work when being run from Octave). Any ideas? I couldn't find any other example (other than `horn_antenna.py`) that imports an STL.

matthuszagh, 2022-04-11:
Hm, I'm not able to reproduce this. As you mentioned, I was only able to see something similar when it wasn't able to find the file.

Do you get any stdout/stderr messages about the stl file not being found?

Is the stl file valid? If you share one I can test.

0xCoto, 2022-04-11:
Here are the stl files in case you can manage to reproduce something: https://github.com/0xCoto/OpenEMS-Notes/tree/stl_files

They should be fine because Octave OpenEMS loads them up without issues, but for some reason pyems fails.

Full script (ignore the poor mesh/boundaries for now):
```py
#!/usr/bin/env python3

import os
import sys
import numpy as np
from pyems.utilities import print_table
from pyems.port import DifferentialMicrostripPort
from pyems.simulation import Simulation
from pyems.field_dump import FieldDump
from pyems.structure import standard_waveguides
from pyems.coordinate import Coordinate3, Box3, Axis
from pyems.mesh import Mesh
from pyems.nf2ff import NF2FF

unit = 1e-3
freq = np.linspace(0e9, 3e9, 501)
sim = Simulation(freq=freq, unit=unit)

# Patch
patch = sim.csx.AddMetal("patch")
stl1 = patch.AddPolyhedronReader(
    filename=os.path.abspath("/home/apostolos/Desktop/Part__Feature001.stl")
)
stl1.ReadFile()

# Substrate
substrate = sim.csx.AddMaterial("substrate", epsilon=2.2, mue=1, kappa=0, sigma=0)
stl2 = substrate.AddPolyhedronReader(
    filename=os.path.abspath("/home/apostolos/Desktop/Part__Feature002.stl")
)
stl2.ReadFile()

# Ground
ground = sim.csx.AddMetal("ground")
stl3 = ground.AddPolyhedronReader(
    filename=os.path.abspath("/home/apostolos/Desktop/Part__Feature.stl")
)
stl3.ReadFile()

port = DifferentialMicrostripPort(
    sim=sim,
    box=Box3(
        Coordinate3(-1, -75 / 2, -1.6),
        Coordinate3(-1, -75 / 2, 0),
    ),
    excitation_axis=Axis("z"),
    propagation_axis=Axis("y"),
    excite=True,
    number=1,
    gap=1.6,
    thickness=0,
)
# port.add_metal_shell(thickness=5)

# Mesh
mesh = Mesh(
    sim=sim,
    metal_res=1 / 20,
    nonmetal_res=1 / 10,
    # smooth=(1.5, 1.5, 1.5),
    min_lines=5,
    expand_bounds=((16, 16), (16, 16), (8, 24)),
)

[... 73 lines cut ...]
  warn(
/home/apostolos/Desktop/oems_tests/pyems/pyems/mesh.py:664: UserWarning: Mesh line at pos 26.3958 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.21. For convenience the last three lines are: 17.8831, 22.5417 and 26.3958.
  warn(
/home/apostolos/Desktop/oems_tests/pyems/pyems/mesh.py:664: UserWarning: Mesh line at pos 42.7753 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.25. For convenience the last three lines are: 34.1042, 37.9583 and 42.7753.
  warn(
/home/apostolos/Desktop/oems_tests/pyems/pyems/mesh.py:664: UserWarning: Mesh line at pos 100.6774 for dimension 1 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.20. For convenience the last three lines are: 84.5697, 91.8899 and 100.6774.
  warn(
/home/apostolos/Desktop/oems_tests/pyems/pyems/mesh.py:664: UserWarning: Mesh line at pos -1.6000 for dimension 2 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.28. For convenience the last three lines are: -1.6000, -1.6000 and -1.6000.
  warn(
/home/apostolos/Desktop/oems_tests/pyems/pyems/mesh.py:664: UserWarning: Mesh line at pos 0.0000 for dimension 2 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.50. For convenience the last three lines are: -0.0157, -0.0063 and 0.0000.
  warn(
/home/apostolos/Desktop/oems_tests/pyems/pyems/mesh.py:664: UserWarning: Mesh line at pos 0.0080 for dimension 2 violates smoothness. Smoothness was set to 1.20 but this line creates a spacing with factor 1.27. For convenience the last three lines are: -0.0063, 0.0000 and 0.0080.
  warn(
CSPrimPolyhedronReader::ReadFromXML: Error, can't read filename!
CSPrimPolyhedronReader::ReadFromXML: Error, can't read filename!
CSPrimPolyhedronReader::ReadFromXML: Error, can't read filename!
VTKPrimitives::AddCube: Warning, can't draw a Point Box... skipping
VTKPrimitives::AddCube: Warning, can't draw a Point Box... skipping
VTKPrimitives::AddCube: Warning, can't draw a Point Box... skipping
VTKPrimitives::AddCube: Warning, can't draw a Point Box... skipping
VTKPrimitives::AddCube: Warning, can't draw a Point Box... skipping
VTKPrimitives::AddCube: Warning, can't draw a Point Box... skipping
Continue simulation (y/n)? n
Terminating simulation.
```

matthuszagh, 2022-04-12:
I can't reproduce this either; works fine on my computer. The only difference in my test is that I put the STL files in /tmp.

The specific error you're triggering is [here](https://github.com/thliebig/CSXCAD/blob/3c4fb665d934349e048fbcc31d6be203483b5e98/src/CSPrimPolyhedronReader.cpp#L92). `QueryStringAttribute` is a tinyxml function. You can download a tarball of the sources for that [here](https://sourceforge.net/projects/tinyxml/files/latest/download). Here is some relevant code from that:

```cpp
	int QueryStringAttribute( const char* name, std::string* _value ) const {
		const char* cstr = Attribute( name );
		if ( cstr ) {
			*_value = std::string( cstr );
			return TIXML_SUCCESS;
		}
		return TIXML_NO_ATTRIBUTE;
	}
```
```cpp
const char* TiXmlElement::Attribute( const char* name ) const
{
	const TiXmlAttribute* node = attributeSet.Find( name );
	if ( node )
		return node->Value();
	return 0;
}
```

Anyway, `Attribute` seems to be returning a null pointer in your case. Unfortunately I'm not really familiar with the tinyxml codebase. The strange thing is it seems to be complaining about `"Filename"`, not the actual file name. My first guess was that you'd used invalid characters in the file name but that doesn't seem to be the case. I'd dig through that tinyxml code if I were you and I'd double-check the installation, in particular with respect to tinyxml.

Does my horn-antenna example work for you? Are you able to run other simulations that don't use the stl reader? It would be a bit strange if tinyxml failed in this specific case but was otherwise able to write XML files fine.

0xCoto, 2022-04-12:
Hmm, looks like I'm getting the same error with the horn example. I'll double-check the installation; perhaps we might have missed something in the [build script](https://github.com/BlitPlatform/compute-servers/blob/main/build_solver.sh).

thomaslepoix, 2022-04-13:
Couldn't it be related to [this](https://github.com/thliebig/CSXCAD/commit/c14eb8441e71f81beb4ab96119be7bd184b3ed7a)? In which case the cause of your problem would be the Debian OpenEMS packages we use that are not up to date (2019/01/03).

matthuszagh, 2022-04-13:
> Couldn't it be related to [this](https://github.com/thliebig/CSXCAD/commit/c14eb8441e71f81beb4ab96119be7bd184b3ed7a)? In which case the cause of your problem would be the Debian OpenEMS packages we use that are not up to date (2019/01/03).

Yep, I'd completely forgotten about that. Could definitely be it.

0xCoto, 2022-04-13:
@thomaslepoix Is there a good reason to use the outdated Debian package at all? Could you update the build script so it grabs the latest version from git, and also takes care of the `./python3-openems*.deb` install? I tried to install the latest version using:
```sh
sudo apt-get install build-essential cmake git libhdf5-dev libvtk7-dev libboost-all-dev libcgal-dev libtinyxml-dev qtbase5-dev libvtk7-qt-dev
sudo apt-get install octave liboctave-dev
sudo pip3 install matplotlib cython h5py
git clone --recursive https://github.com/thliebig/openEMS-Project.git
cd openEMS-Project
./update_openEMS.sh ~/opt/openEMS
./update_openEMS.sh ~/opt/openEMS --python
```
but got the initial error again after trying pyems: `No module named 'CSXCAD'`. I'm guessing we have to insert something along the lines of `sudo apt install python3-openems*.deb` like we did last time, but for the latest version?

thomaslepoix, 2022-04-14:
You can't really mix manual and APT based installation.

- If you want it immediately, I would suggest to follow the procedure you spotted [here](https://github.com/matthuszagh/pyems/issues/3).

- Using the `.deb` is IMHO better because of the usual benefits of APT and because the package contains patches about various oddities. I could try a script that updates on the fly the existing package before regenerating it. Will also submit my patch about build flags and request an update from Debian maintainer.

0xCoto, 2022-04-14:
> I could try a script that updates on the fly the existing package before regenerating it.

That would be perfect, thanks! We're currently investigating some methods of achieving speedup in the FDTD code, so being able to update from a git repository whenever needed would be ideal.

0xCoto, 2022-05-16:
Solved - Thomas also helped built a convenient [script](https://github.com/BlitPlatform/compute-servers/blob/main/build_solver.sh) for installing openEMS on fresh debian installations. The relevant [PR](https://github.com/thliebig/CSXCAD/pull/26) for CSXCAD fixed the problem. It's also important to make sure the submodules are up-to-date; we cloned using `--recursive --remote-submodules` to ensure the submodules are updated in case Thorsten has forgotten to link them to the latest commits (e.g. the CSXCAD submodule is not currently linked to the latest commit).

Will open another issue for pyems usage questions, cheers!

closed 2022-05-16 by 0xCoto