matthuszagh/pyems #61: Taper is half implemented

BatchDrake, 2023-02-07:
Taper implementation is incomplete. According to the code (https://github.com/matthuszagh/pyems/blob/f63fcf85bf2e8e26ff685035b5d36b4a85860268/pyems/structure.py#L1815) the polygons themselves are being created, but they are not being appended to the polygon list.

matthuszagh, 2023-02-07:
I think I see what you're saying, but it would be helpful if you could provide a reproducible example that shows incorrect behavior. It's been a couple years since I wrote this code, so that would help me be sure I'm changing things in the intended way.

BatchDrake, 2023-03-03:
Hey, sorry for the late answer, been busy tweaking simulations here and there. Here's the reproducer: it is basically two microstrips with a taper in the middle you can conditionally disable. The problem arises when you attempt to save the footprint: as the Taper does not append these polygons to the polygon list, it breaks with an exception:

```python

#!/usr/bin/env python3

#
# Set these to False to disable addition of the taper section and
# generation of the footprint respectively
#

ADD_TAPER       = True
WRITE_FOOTPRINT = True

import os
import sys
import numpy as np
from pyems.structure import Microstrip, PCB, MicrostripCoupler, Taper
from pyems.simulation import Simulation
from pyems.pcb import common_pcbs
from pyems.calc import phase_shift_length, microstrip_effective_dielectric
from pyems.utilities import print_table
from pyems.coordinate import Coordinate2, Axis, Box3, Coordinate3
from pyems.mesh import Mesh
from pyems.field_dump import FieldDump, DumpType
from pyems.kicad import write_footprint_set

freq = np.linspace(0e9, 18e9, 501)
ref_freq = 5.6e9
unit = 1e-3
sim = Simulation(freq=freq, unit=unit, reference_frequency=ref_freq)

pcb_prop = common_pcbs["oshpark4"]
trace_width = 0.38
trace_spacing = 0.2
eeff = microstrip_effective_dielectric(
    pcb_prop.substrate.epsr_at_freq(ref_freq),
    substrate_height=pcb_prop.substrate_thickness(0, unit=unit),
    trace_width=trace_width,
)
coupler_length = phase_shift_length(90, eeff, ref_freq)
pcb_len = 2 * coupler_length
pcb_width = 0.5 * pcb_len
pcb = PCB(
    sim=sim,
    pcb_prop=pcb_prop,
    length=pcb_len,
    width=pcb_width,
    layers=range(3),
    omit_copper=[0],
)

structures = []

port0_x = 0
port0_y = 0

s1 = Microstrip(
    pcb=pcb,
    position=Coordinate2(np.average([port0_x, -pcb_len / 2]), port0_y),
    length=port0_x + pcb_len / 2,
    width=trace_width,
    propagation_axis=Axis("x"),
    port_number=1,
    excite=True,
    ref_impedance=50,
    feed_shift=0.3,
)

structures.append(s1)

port1_x = 0
port1_y = 0
s2 = Microstrip(
    pcb=pcb,
    position=Coordinate2(np.average([port1_x, pcb_len / 2]), port0_y),
    length=pcb_len / 2 - port1_x,
    width=trace_width,
    propagation_axis=Axis("x", direction=-1),
    port_number=2,
    excite=False,
    ref_impedance=50,
)

structures.append(s2)

if ADD_TAPER:
    taper_len = pcb_width / 3
    taper = Taper(
        pcb = pcb,
        position = Coordinate2(taper_len / 2, 0),
        pcb_layer = 0,
        width1 = trace_width,
        width2 = trace_width * 4,
        length = taper_len,
        rotation = 90),

    structures.append(taper)
    
Mesh(
    sim=sim,
    metal_res=1 / 120,
    nonmetal_res=1 / 40,
    smooth=(1.1, 1.5, 1.5),
    min_lines=3,
    expand_bounds=((0, 0), (0, 0), (10, 20)),
)

FieldDump(
    sim=sim,
    box=Box3(
        Coordinate3(-pcb_len / 2, -pcb_width / 2, 0),
        Coordinate3(pcb_len / 2, pcb_width / 2, 0),
    ),
    dump_type=DumpType.current_density_time,
)

if WRITE_FOOTPRINT:
    write_footprint_set(structures, "coupler_20db", "coupler_20db.kicad_mod")

sim.run(csx = True)
```

matthuszagh, 2023-03-09:
This should now be fixed in ff563e8ba85e14b3c5a9a2adfae228f4910d3a7e. Feel free to reopen if you run into any issues.

closed 2023-03-09 by matthuszagh