matthuszagh/pyems #64: example does not work

ninjamongeese, 2023-09-22:
While generating mesh, i get the following error:

```
C:\ProgramData\anaconda3\lib\site-packages\scipy\optimize\_minpack_py.py:178: RuntimeWarning: The iteration is not making good progress, as measured by the 
  improvement from the last five Jacobian evaluations.
  warnings.warn(msg, RuntimeWarning)
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In[8], line 1
----> 1 Mesh(
      2     sim=sim,
      3     metal_res=1 / 80,
      4     nonmetal_res=1 / 40,
      5     smooth=(1.1, 1.5, 1.5),
      6     min_lines=5,
      7     expand_bounds=((0, 0), (0, 0), (10, 40)),
      8 )

File C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:594, in Mesh.__init__(self, sim, metal_res, nonmetal_res, smooth, min_lines, expand_bounds, simulation_bounds)
    591 # set later
    592 self.bounded_types = None
--> 594 self.generate_mesh()

File C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:619, in Mesh.generate_mesh(self, show_pml)
    617 self._gen_mesh_for_bounded_types(self.size_ordered_bounded_types)
    618 self._trim_air_mesh()
--> 619 self._smooth_pml_mesh_lines()
    621 self._set_mesh_from_lines()
    622 if show_pml:

File C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:722, in Mesh._smooth_pml_mesh_lines(self)
    717     spacing = limit_spacing * smooth_factor
    718 elif (
    719     spacing < limit_spacing
    720     and limit_spacing / spacing >= smooth_factor
    721 ):
--> 722     raise RuntimeError(
    723         "PML mesh smoothing is increasing the simulation "
    724         "box size.  This should never happen.  Either "
    725         "structures in your PML are non-uniform in the "
    726         "direction of the PML in which case you'll get an "
    727         "error anyway, or this is a bug in pyems."
    728     )
    730 # only smooth lines if necessary
    731 if np.isclose(spacing, limit_spacing, rtol=1e-3, atol=0):

RuntimeError: PML mesh smoothing is increasing the simulation box size.  This should never happen.  Either structures in your PML are non-uniform in the direction of the PML in which case you'll get an error anyway, or this is a bug in pyems.
```

ninjamongeese, 2023-09-22:
3.10.9 | packaged by Anaconda, Inc. | (main, Mar  1 2023, 18:18:15) [MSC v.1916 64 bit (AMD64)]

U:\>conda list
# packages in environment at C:\Users\All Users\anaconda3:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12             pyhd3eb1b0_0
anaconda-anon-usage       0.4.0           py310h8ddfbff_0
anaconda-client           1.11.2          py310haa95532_0
anaconda-navigator        2.4.3           py310haa95532_0
anaconda-project          0.11.1          py310haa95532_0
anyio                     3.5.0           py310haa95532_0
appdirs                   1.4.4              pyhd3eb1b0_0
argon2-cffi               21.3.0             pyhd3eb1b0_0
argon2-cffi-bindings      21.2.0          py310h2bbff1b_0
arrow                     1.2.3           py310haa95532_1
astroid                   2.14.2          py310haa95532_0
astropy                   5.1             py310h9128911_0
asttokens                 2.0.5              pyhd3eb1b0_0
atomicwrites              1.4.0                      py_0
attrs                     22.1.0          py310haa95532_0
automat                   20.2.0                     py_0
autopep8                  1.6.0              pyhd3eb1b0_1
babel                     2.11.0          py310haa95532_0
backcall                  0.2.0              pyhd3eb1b0_0
backports                 1.1                pyhd3eb1b0_0
backports.functools_lru_cache 1.6.4              pyhd3eb1b0_0
backports.tempfile        1.0                pyhd3eb1b0_1
backports.weakref         1.0.post1                  py_1
bcrypt                    3.2.0           py310h2bbff1b_1
beautifulsoup4            4.11.1          py310haa95532_0
binaryornot               0.4.4              pyhd3eb1b0_1
black                     22.6.0          py310haa95532_0
blas                      1.0                         mkl
bleach                    4.1.0              pyhd3eb1b0_0
blosc                     1.21.3               h6c2663c_0
bokeh                     2.4.3           py310haa95532_0
boltons                   23.0.0          py310haa95532_0
bottleneck                1.3.5           py310h9128911_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
brotlipy                  0.7.0           py310h2bbff1b_1002
bzip2                     1.0.8                he774522_0
ca-certificates           2023.08.22           haa95532_0
certifi                   2023.7.22       py310haa95532_0
cffi                      1.15.1          py310h2bbff1b_3
cfitsio                   3.470                h2bbff1b_7
chardet                   4.0.0           py310haa95532_1003
charls                    2.2.0                h6c2663c_0
charset-normalizer        2.0.4              pyhd3eb1b0_0
click                     8.0.4           py310haa95532_0
cloudpickle               2.0.0              pyhd3eb1b0_0
clyent                    1.2.2           py310haa95532_1
colorama                  0.4.6           py310haa95532_0
colorcet                  3.0.1           py310haa95532_0
comm                      0.1.2           py310haa95532_0
conda                     23.3.1          py310haa95532_0
conda-build               3.24.0          py310haa95532_0
conda-content-trust       0.1.3           py310haa95532_0
conda-pack                0.6.0              pyhd3eb1b0_0
conda-package-handling    2.0.2           py310haa95532_0
conda-package-streaming   0.7.0           py310haa95532_0
conda-repo-cli            1.0.41          py310haa95532_0
conda-token               0.4.0              pyhd3eb1b0_0
conda-verify              3.4.2                      py_1
console_shortcut          0.1.1                         4
constantly                15.1.0          py310haa95532_0
contourpy                 1.0.5           py310h59b6b97_0
cookiecutter              1.7.3              pyhd3eb1b0_0
cryptography              39.0.1          py310h21b164f_0
[... 334 lines cut ...]
wcwidth                   0.2.5              pyhd3eb1b0_0
webencodings              0.5.1           py310haa95532_1
websocket-client          0.58.0          py310haa95532_4
werkzeug                  2.2.2           py310haa95532_0
whatthepatch              1.0.2           py310haa95532_0
wheel                     0.38.4          py310haa95532_0
widgetsnbextension        3.5.2           py310haa95532_0
win_inet_pton             1.1.0           py310haa95532_0
wincertstore              0.2             py310haa95532_2
winpty                    0.4.3                         4
wrapt                     1.14.1          py310h2bbff1b_0
xarray                    2022.11.0       py310haa95532_0
xlwings                   0.29.1          py310haa95532_0
xz                        5.2.10               h8cc25b3_1
yaml                      0.2.5                he774522_0
yapf                      0.31.0             pyhd3eb1b0_0
zeromq                    4.3.4                hd77b12b_0
zfp                       0.5.5                hd77b12b_6
zict                      2.1.0           py310haa95532_0
zipp                      3.11.0          py310haa95532_0
zlib                      1.2.13               h8cc25b3_0
zope                      1.0             py310haa95532_1
zope.interface            5.4.0           py310h2bbff1b_0
zstandard                 0.19.0          py310h2bbff1b_0
zstd                      1.5.2                h19a0ad4_0

matthuszagh, 2023-09-22:
Please specify the example you're running.

ninjamongeese, 2023-09-22:
The front page example. I guess looking at the code page its actually labelled "Tutorial". Hope that helps!

This error seems to be related to the argument 
`Mesh(expand_bounds=((0, 0), (0, 0), (10, 40)))`

I took out that argument entirely as a test, as I'm unsure what maximum Z height is of the simulation for this configuration. When I take out just that argument to let it use the default parameters, i was able to get that step to run. It generated a mesh but  with some warnings. Not sure if its related or even an issue. warnings below:

```
C:\ProgramData\anaconda3\lib\site-packages\scipy\optimize\_minpack_py.py:178: RuntimeWarning: The iteration is not making good progress, as measured by the 
  improvement from the last five Jacobian evaluations.
  warnings.warn(msg, RuntimeWarning)
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos -5.8817 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.12. For convenience the last three lines are: -6.2229, -6.0423 and -5.8817.
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos -5.5787 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.13. For convenience the last three lines are: -5.8817, -5.7212 and -5.5787.
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos -4.3335 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.15. For convenience the last three lines are: -4.4190, -4.3732 and -4.3335.
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos -3.9962 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.28. For convenience the last three lines are: -4.0150, -4.0045 and -3.9962.
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos -3.9614 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.24. For convenience the last three lines are: -3.9798, -3.9716 and -3.9614.
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos 3.9798 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.24. For convenience the last three lines are: 3.9614, 3.9716 and 3.9798.
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos 4.0150 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.28. For convenience the last three lines are: 3.9962, 4.0045 and 4.0150.
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos 4.4190 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.15. For convenience the last three lines are: 4.3335, 4.3732 and 4.4190.
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos 5.8817 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.13. For convenience the last three lines are: 5.5787, 5.7212 and 5.8817.
C:\ProgramData\anaconda3\lib\site-packages\pyems-0.1.0-py3.10.egg\pyems\mesh.py:664: UserWarning: Mesh line at pos 6.2229 for dimension 0 violates smoothness. Smoothness was set to 1.10 but this line creates a spacing with factor 1.12. For convenience the last three lines are: 5.8817, 6.0423 and 6.2229.
```

matthuszagh, 2023-09-22:
Got it. Thanks for the bug report I'll take a look in a couple days (can't this weekend).

Those warnings aren't necessarily a problem; they merely indicate that the mesher can’t meet the constraints you've set (or I suppose the ones I set, since it's the tutorial). When you're writing your own simulations, you'll find this crop up and often it means you just were a bit too strict with the mesh requirements and the mesh is still good. If you deem the mesh result inadequate, though, you'll have to adjust the design in some respect so the requirements can be met.