matthuszagh/pyems #67: Research citations

keixthb, 2025-03-19:
Hey @matthuszagh, do you have a preferred way to cite the pyems library in bibtex? I didn't see anything in your readme so I thought I'd ask. Thanks!

matthuszagh, 2025-03-20:
I'm glad to hear pyems will be used as part of a research paper! I actually don't have strong opinions on this and am open to suggestions, but perhaps something in line with the recommendation made [here](https://academia.stackexchange.com/a/14015)? So, something like

```
@misc{Huszagh2023,
  author = {Huszagh, Matt},
  title = {pyems},
  year = {2023},
  publisher = {GitHub},
  journal = {GitHub repository},
  howpublished = {\url{https://github.com/matthuszagh/pyems}},
  commit = {79ffdc50f8039b9e58814aed64d77e87c57d1677}
}
```

Please don't forget to acknowledge openems too.

keixthb, 2025-03-21:
Ok cool thanks. Yes, I cited openEMS as well. I like the idea of including the commit hash.

closed 2025-03-21 by keixthb